Domesday86/DomesdayDuplicator #9: Capture of lead-out using player integrated capture mode

simoninns, 2018-04-27:
The Linux front-end application provides automated capture of a laserdisc including lead-in by beginning sampling before the disc is spun up.  When the disc hits the last frame the player pauses.  It's questionable if there is lead-out information which isn't seen at the end of the sample.

Testing in normal capture mode would continue sampling after the disc ends.   Captures should be made in both modes and compared to see if this is a potential issue (as a normal capture wouldn't be affected).

happycube, 2018-05-16:
IIRC the end of side signal is either specified in the lead in or the VBI data for the last frame itself.  Generally on a player set to repeat a side, it'll rapidly scan through the disk backwards.

simoninns, 2018-08-19:
This doesn't seem to be a problem - so I'm closing it.

closed 2018-08-19 by simoninns