Domesday86/DomesdayDuplicator #13: CLD-V2800 support info

happycube, 2018-05-20:
I got PIC (which is quite nifty!) working with my V2800 using this info:  Player ID is P153701, and 4800bps is a possible baud rate which I had to hack the GUI for (the dip switch can be set for 9600, but there's a parameter setting I remember using for which 4800 is the limit)

simoninns, 2018-05-20:
I'd figured that it would work with other Pioneer players; so nice to know it does :)  My problem (with adding support to the code-base) is testing.  But I will try to add the player's ID to the existing base soon (since it's confirmed working).

Glad you like the PIC feature - it does make capturing a lot easier!

simoninns, 2018-06-30:
There seems to be an intermittent issue when connecting to the V2800.  Some times the GUI responds: 

"sendSerialCommand(): Received error response from player: "E04\r"
smConnectingState(): Player ID is unknown -  "E04\r"
smSerialErrorState(): In serial error state; closed serial port"

simoninns, 2018-07-07:
The plan here is to support multiple retries of player identification commands.  This will allow for players that have an unknown state or simply don't reply correctly the first time.

simoninns, 2018-08-12:
Multiple retries of all serial communication is now included in the latest commit 2341d4d.  I'm going to close this issue since it's very general - feel free to open new issues for more specific issues around player communication.

closed 2018-08-12 by simoninns