Domesday86/DomesdayDuplicator #14: Test PIC function with a disc that pauses automatically

simoninns, 2018-05-20:
There is an edge-case when capturing discs automatically using PIC.  Some discs (like the GGV1069) have a pause command included in a frame; this may cause the player to pause during PIC and the capture state-machine doesn't deal with this (potentially continuing forever recording the paused frame).

happycube, 2018-05-20:
The V2800 and V5000 have serial control for Picture Stop Cancel - command is "1PS": https://www.pioneerelectronics.com/pio/pe/images/portal/cit_3424/31301223PIB152007_V28_Notes.pdf

simoninns, 2018-05-20:
Great info! Thanks for the reference; that should solve the issue nicely.

happycube, 2018-05-20:
... unfortunately, it only solves the issue for those players officially.  It's conceivable that later-run (1996+) LD-V4300D/4400's and V8000's got a firmware update for it.

simoninns, 2018-06-02:
The only reference I can find for this in the 4300 documentation is the E15 "Picture Stop" error code.  Might be possible to use that, but needs more investigation.

simoninns, 2018-06-02:
I've just tested a PIC capture on a PAL disc that has a pause in one of the frames.  As soon as the player pauses the PIC state-machine sees this as the same result as the last target frame being reached.  Therefore, without error, it stops the capture cleanly (but not at the end frame specified by the user).