Domesday86/DomesdayDuplicator #15: Adjustable gain for DD board

happycube, 2018-05-21:
My DD 2.0 with a 3dB attenuator is fine for 8" CAV disks and 8/12" CLV, but I was trying to capture THX Wow! and A Video Standard's signals and they were too strong causing interference patterns etc.  It would be nice to have an adjustment on the board I could use, or even better (but far more difficult!) would be an electronic gain control controlled by a DAC on the board.

simoninns, 2018-05-21:
You can modify the gain of the RF front-end according to your player by changing the values of two resistors on the Duplicator board. This is covered in the documentation;

Board 2_2: https://www.domesday86.com/?page_id=1836#Modifying_the_RF_front-end_gain
Board 2_0: https://www.domesday86.com/?page_id=1066#Modifying_the_RF_front-end_gain

If you are not using the reference player, then you will need to measure the RF gain of the player (over the whole disc) using an oscilloscope and perform the documented calculations (leaving some 'headroom' for peaking of the signal).  It's best to do this by connecting the player to the duplicator and measuring the input on pin 3 of the OPA690 IC (as this negates the need to consider impedance).

I looked into both AGC and programmable gain in the design, but came to the conclusion that the additional signal distortion caused by the circuitry was not optimal considering the use-case is typically use with a single LDVP model (and selecting/changing two resistors was a relatively easy modification).  The RF front-end is the most sensitive part of the design - so simple is better.

I've now added additional information to the 2_2 board hardware guide covering the recommended method of measurement.

simoninns, 2018-05-21:
The chip is rated for 5V, but the ADC input maximum is 3.5V (however this is the ADC range, the datasheet doesn't state a pin maximum tolerance).  I don't think anything less than 5Vs would damage the chip though, so a little testing should be fine.

The fact that later players don't have a level adjustment is actually a benefit, since the calibration can't be off :)

Taking a 0805 resistor off the board is actually quite easy.  Touch one side with the soldering iron (for second) and then place the tip on the other side and hold; the component is so small that the solder on both sides melt - then you just pluck it off quickly with some tweezers.  Clean the pads with desoldering braid and then add a small amount of solder to just one pad.  Heat the soldered pad and pop in the new resistor using tweezers.  Then just solder the other side; job done!

happycube, 2018-05-22:
Cool :)  I'll probably stick with the attenuators (assuming 6 and 10 work as well as 3 did), since it makes handling multiple disk types easier (12" CAV runs ~2x as high as a CLV/CAA disk!)

happycube, 2018-05-27:
I managed to get a 560ohm resistor on the board - and 3x gain was _still_ too high for the V2800 at the end of AVS.  I still think it would be nice to have some sort of gain selection (either a potentiometer or even jumpers) but I can see how that's beyond the main project scope ;)

simoninns, 2018-05-27:
Actually, the idea of providing a few preset gains is more of a possiblilty; I'd need to look into what happens if you use a DIP switch or headers (as they can act as antenna for RF) - but it's worth considering... What preset gains do you think would be useful?

simoninns, 2018-06-19:
Fixed in commit c3b998e - New board design now supports user-adjustable gain from x2 to x8.5 configurable using an on-board DIP switch

closed 2018-06-19 by simoninns