Domesday86/DomesdayDuplicator #21: Low-pass anti-alias filter issue

simoninns, 2018-06-05:
Testing (by decoding the RF sample using ld-decode) has shown an issue with the current low-pass filter design.

On NTSC CAV discs the white output is at 9.35MHz which causes a harmonic at 28.05MHz.  This harmonic is aliased into the captured signal at 11.95MHz when capturing at 40MSPS.

The aliasing causes distortion of the video signal after around frame 40,000 of a NTSC CAV disc and causes signal degradation which is particularly bad at frames 50,000 and beyond.  Reproduction of the issue requires a 12" test disc (which are difficult to source), so this was not see in the initial testing of the 2_2 board.

There are three possible solutions so far:

1) Modify the current LPF to provide a lower-frequency stop-band
2) Increase the sampling rate to capture the harmonic
3) Redesign the filter to be active (i.e. increase the 'drop-off' effect of the filter to remove the harmonic from the sampled signal)

(1) is the simplest solution, but has the side-effect of attenuating the 'good' signal too much making decoding more difficult

(2) is practical, but requires a more expensive ADC (sampling would need to be 60MSPS).  This is really just 'masking' the issue at the cost of higher sampling rates (and generation of unnecessary large RF samples)

(3) is probably the correct long-term fix - but requires a redesign of the duplicator board

happycube, 2018-06-05:
The data you sent me for 1) looks good (and remember that film-based CAV's frame count is 4/5'ths of the disk frame - so a film sample at 41700 is ~52000, which is enough to clearly show the harmonic)

2) It's not a big clock bump (44mhz) to move the harmonic alias out of the video RF decoding - but it still requires a 60mhz ADC and it's associated cost.

There's also option 4, which is to filter out the offending alias in software, but it really hurts color saturation and is best avoided if possible.

simoninns, 2018-06-05:
Ok, so the recommended solution for the 2_0/2_2 boards is to replace C26 (a 220pF capacitor) with a 330pF capacitor.  This shifts the LPF stop-band down from 15MHz to 10Mhz and greatly reduces (but doesn't completely remove) the unwanted harmonic.

The real solution is to increase the order of the low-pass filter.  Unfortunately, with a passive RC filter, I think this is going to cause too much attenuation of the whole signal.  A better choice would be to use some op amps to form a butterworth filter in a sallen-key low-pass configuration (with a gain of 1).  Ideally the filter should push the signal below the noise-floor before hitting 20MHz.  A second or fourth order filter should do the trick, but I need to look at the detail of the design to be sure.

happycube, 2018-06-05:
I'll have 330 (and 270) pF caps in a few days to try out on my board.  I still need to get the 40mhz TBC written and working before I can get a real idea about the new filter's effect on chroma noise, but w/330 the white SNR is much better.

simoninns, 2018-06-06:
The main issue seems to be related to the design of the low-pass filter on the 2_2 board; it does not slope correctly and seems to flatten off after about 20MHz (causing the high-frequency signals to be stronger than they should be).  Right now it seems a 3 component LRC passive filter design would provide much better Q.  The optimum filter settings are somewhere between about 13MHz and 14MHz -3dB stop-band.  Needs some breadboarding and testing to confirm.

simoninns, 2018-06-19:
Fixed in commit c3b998e - Elliptic 2nd order filter now included in design

closed 2018-06-19 by simoninns