Domesday86/DomesdayDuplicator #24: Bit-packed output files

simoninns, 2018-06-30:
Since the output from the ADC is 10 bit, it would be nice to have the option of saving the Duplicator output in a 'packed' format to save disc space (rather than just 16-bit signed data).

happycube, 2018-07-01:
Thinking about it, it would be ideal to store the packed bits in unsigned format - if signed format is required it's easy to do that in the unpacking process.  I updated dd[un]pack.c in my ld-decode git just now to do this.

simoninns, 2018-07-01:
My plan was 10-bit unsigned, so I think we are in alignment :)

simoninns, 2018-07-04:
In order to lower the USB throughput, the 10-bit packing needs to be performed by the FPGA on the ADC input data.  This means that, before the DCFIFO input, the data should be packed and a new asynchronous 'clock' generated to control the FIFO writes.  The DCFIFO will therefore need to be changed to 16-bit wide rather than 10-bits.  Making the change on the write-side of the DCFIFO prevents the change from requiring a rewrite of the FPGA->FX3 data transfer state-machine model.

This change will also 'break' the test data functions in the GUI as the expected data will change.  The GUI will need to unpack the data to retrieve the original 10-bit values for comparison.

In order to test the resulting samples a small utility for reformatting the 10-bit packed samples files back into 16-bit signed values will also need to be created.

simoninns, 2018-07-07:
There's an issue with keeping the bit-stream in sync using 80-bit packed data from the FPGA.  To save over-complicating the FPGA<->FX3 code I'm planning on removing the 16-bit signed data conversion from the FPGA code and passing the raw 10-bit unsigned samples instead.  Then the GUI can pack the data into 80-bit packets (5x16-bit words = 8x 10-bit samples).  This (unfortunately) won't lower the USB communication bandwidth, but it will give reduced RF sample file sizes (making the files about 40% smaller).

simoninns, 2018-08-18:
10-bit packed byte streaming is supported in commit 18f1c75.  Data is transferred as unsigned 10-bit over the USB 16-bit endpoint and is then converted and packed by the GUI application (this is optional, you can also select the previous 16-bit sample format too).

closed 2018-08-18 by simoninns