Domesday86/DomesdayDuplicator #25: Ability to monitor amplitude of captured RF

simoninns, 2018-06-30:
Since the Duplicator now has variable gain, it would be useful to be able to monitor the amplitude of the RF capture in real-time from the GUI.

simoninns, 2018-08-19:
Just a note to say that the amplitude should be represented by a rolling average based on RMS with the output on a logarithmic decibel scale.

wondras, 2018-09-11:
I was going to suggest this very feature, but you of course beat me to it. :)

A few additional ideas:
- If the amplitude measurement or display impacts performance during capture, it could be implemented as a separate "signal check" mode/screen.
- A "level" meter with a resettable clipping indicator would make the amplitude easier to read/interpret at a glance (how many clipped samples per second would be considered 'clipping')
- A waveform preview would give more of a "feel" for what's happening, but perhaps it's not as useful as a meter or numeric display.

simoninns, 2018-09-23:
The main issue with implementing a VU meter is the load on the system - at the moment the USB transfers are pushed directly into the disk buffers using fast memory transfer.  Analysing the data would require slow iteration.  An alternative is to provide feedback on the amplitude average of each disk buffer since the data is read anyway due to 10- and 16-bit conversion.

This would give feedback that was only updated every second or so (in the form of a text-based label), but it would probably serve the intended purpose.

TokugawaHeavyIndustries, 2022-09-09:
#130

simoninns, 2022-12-17:
PR accepted, issue closed.

closed 2022-12-17 by simoninns