Domesday86/DomesdayDuplicator #37: Support for Windows OS

simoninns, 2018-08-25:
Currently the Duplicator application is only compatible with Ubuntu/Linux.  It should be possible (using QT) to make the capture application cross-platform.

simoninns, 2018-09-23:
The new GUI rewrite goes a long way towards this due to the streamlined USB class implementation.  There is still the issue of libusb on windows not supporting hotplug.  Also there needs to be a way to install drivers on windows for libusb (I tried zdiag, but it seems to have a very bad affect on a number of library calls).

simoninns, 2018-09-23:
After commit  af09b16 the application is reported to compile on Mac OS cleanly; however attempting to capture causes the same libUSB error found on Windows (with the exception that hotplug is supported on Mac OS).  The primary debug response is: UsbCapture::run(): Transfer launch 0 failed with error: LIBUSB_ERROR_NOT_FOUND

wondras, 2018-09-23:
According to https://github.com/tessel/node-usb/issues/61 on macOS you have to 'claim' the interface before you can transfer data. The solution posted here involves using internal methods, but there is a proper libusb_claim_interface() method that accomplishes the same thing.

Adding this line just after the call to libusb_open() makes the data transfers function correctly:
`responseCode = libusb_claim_interface(usbDeviceHandle, 0);`

simoninns, 2018-09-24:
Seems like the configuration mechanism isn't functioning under Windows either.

simoninns, 2018-09-24:
All fixed apart from device detach detection - I'll open a new issue for that though to stop things from being confusing

commit d0d1054b (2018-09-24): Added device state polling for Windows (where hot-plug events are not supported).  Fixes #37
closed 2018-09-24 by simoninns