Domesday86/DomesdayDuplicator #43: Text clipping in UI

wondras, 2018-09-23:
Some labels/captions in the UI are being clipped, with somewhat different cicrumstances in both Ubuntu and macOS.

I am not aware of any special display or scaling settings on my ubuntu setup, though it is running in a VM. The Mac is an early 2013 Retina MacBook Pro with "best for retina" scaling option.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

wondras, 2018-09-23:
Also noted that the middle radio button in the auto-capture dialog has a large outline around it, who knows why...

simoninns, 2018-09-24:
> Also noted that the middle radio button in the auto-capture dialog has a large outline around it, who knows why...

That's the highlight for keyboard control (part of the OS) - the field tab ordering is incorrect, which is why it highlights the middle radio button rather than the top as you'd expect.

simoninns, 2018-09-24:
Should all be looking much better in commit  b77ca87 - Please take a look at the mac and ubuntu GUIs and let me know if there are still issues

wondras, 2018-09-24:
Tests good on macOS and Ubuntu.  As a side benefit, the UI seems to have the right amount of "air" in it, so it doesn't feel cramped or enormous.

closed 2018-09-24 by wondras