Domesday86/DomesdayDuplicator #44: macOS: DomesdayDuplicator crash when saving preferences

wondras, 2018-09-23:
(Using macOS 10.12.6)
1. Open preferences dialog
2. Click 'Save'

Program exits, and crash report dialog appears with the following details:

Process:               DomesdayDuplicator [16003]
Path:                  /Users/USER/*/DomesdayDuplicator.app/Contents/MacOS/DomesdayDuplicator
Identifier:            Test.DomesdayDuplicator
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           DomesdayDuplicator [16003]
User ID:               501

Date/Time:             2018-09-23 18:46:24.940 -0400
OS Version:            Mac OS X 10.12.6 (16G1510)
Report Version:        12
Anonymous UUID:        751163B1-9535-E8F1-5662-AD7C970E78F0

Sleep/Wake UUID:       1E52CB42-B772-4EC3-AF1B-D09528B95E65

Time Awake Since Boot: 190000 seconds
Time Since Wake:       20000 seconds

System Integrity Protection: enabled

**Crashed Thread:        1  PlayerControl**

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0xffff9f810dfd4b75
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0xffff9f810dfd4b75:
--> shared memory          00007ffffff32000-00007ffffff33000 [    4K] r-x/r-x SM=SHM  
    

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fffd6bf634a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fffd6bf5797 mach_msg + 55
2   com.apple.CoreFoundation      	0x00007fffc0efe874 __CFRunLoopServiceMachPort + 212
3   com.apple.CoreFoundation      	0x00007fffc0efdcf1 __CFRunLoopRun + 1361
4   com.apple.CoreFoundation      	0x00007fffc0efd544 CFRunLoopRunSpecific + 420
5   com.apple.HIToolbox           	0x00007fffc045cebc RunCurrentEventLoopInMode + 240
6   com.apple.HIToolbox           	0x00007fffc045ccf1 ReceiveNextEventCommon + 432
7   com.apple.HIToolbox           	0x00007fffc045cb26 _BlockUntilNextEventMatchingListInModeWithFilter + 71
8   com.apple.AppKit              	0x00007fffbe9f3a54 _DPSNextEvent + 1120
9   com.apple.AppKit              	0x00007fffbf16f7ee -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 2796
10  com.apple.AppKit              	0x00007fffbe9e83db -[NSApplication run] + 926
11  libqcocoa.dylib               	0x0000000110c93e0d 0x110c69000 + 175629
12  org.qt-project.QtCore         	0x000000010eed7d3e QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 398
13  org.qt-project.QtCore         	0x000000010eedc8b1 QCoreApplication::exec() + 369
14  Test.DomesdayDuplicator       	0x000000010e1c4416 main + 70
15  libdyld.dylib                 	0x00007fffd6acf235 start + 1

Thread 1 Crashed:: PlayerControl
0   Test.DomesdayDuplicator       	0x000000010e1e0b33 PlayerCommunication::disconnect() + 99
1   Test.DomesdayDuplicator       	0x000000010e1e4858 PlayerControl::run() + 536
2   org.qt-project.QtCore         	0x000000010ed2caae 0x10ecff000 + 187054
3   libsystem_pthread.dylib       	0x00007fffd6ce893b _pthread_body + 180
4   libsystem_pthread.dylib       	0x00007fffd6ce8887 _pthread_start + 286
5   libsystem_pthread.dylib       	0x00007fffd6ce808d thread_start + 13

Thread 2:: org.libusb.device-hotplug
0   libsystem_kernel.dylib        	0x00007fffd6bf634a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fffd6bf5797 mach_msg + 55
[... 447 lines cut ...]
shared memory                     16.4M       15 
===========                     =======  ======= 
TOTAL                            879.7M     1088 

Model: MacBookPro10,1, BootROM MBP101.00F6.B00, 4 processors, Intel Core i7, 2.8 GHz, 16 GB, SMC 2.3f36
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Graphics: NVIDIA GeForce GT 650M, NVIDIA GeForce GT 650M, PCIe, 1024 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434314753364D465238432D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434314753364D465238432D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xEF), Broadcom BCM43xx 1.0 (7.21.171.133.1a2)
Bluetooth: Version 5.0.5f3, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM768E, 751.28 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 3.0 Bus
USB Device: Domesday Duplicator
Thunderbolt Bus: MacBook Pro, Apple Inc., 23.4

simoninns, 2018-09-24:
Should be fixed in commit  b77ca87 - let me know if it works and I will close this issue

wondras, 2018-09-24:
It works now on Ubuntu and macOS.

simoninns, 2018-09-24:
Great, issue closed then :)

closed 2018-09-24 by simoninns