Domesday86/DomesdayDuplicator #45: macOS: ten second freeze after stopping a capture

wondras, 2018-09-23:
1. Start capture
2. Stop capture

The UI becomes unresponsive (with pinwheel/beach-ball) for about 10 seconds, then functions normally.
 
When run from Terminal, the delay occurs between these two debug messages:

```
UsbCapture::runDiskBuffers(): Thread stopped
UsbCapture::freeDiskBuffers(): Freeing disk buffer memory
```

simoninns, 2018-09-24:
Should be fixed in commit  b77ca87 - please retry and let me know if I can close this issue

wondras, 2018-09-24:
still has the delay... console output is:

`MainWindow::on_capturePushButton_clicked(): Starting capture to file: "/Users/warren/TestData_2018-09-24_12-29-46.lds"
MainWindow::on_capturePushButton_clicked(): Starting transfer
UsbDevice::startCapture(): Starting capture
UsbDevice::startCapture(): Sending start capture USB vendor specific command
UsbDevice::startCapture(): Opening the capture device
UsbDevice::startCapture(): Creating the capture object
UsbDevice::startCapture(): Starting capture process with start()
MainWindow::on_capturePushButton_clicked(): Transfer started
UsbCapture::run(): Setting up the transfers
UsbCapture::allocateDiskBuffers(): Allocating memory for disk buffers
UsbCapture::runDiskBuffers(): Thread started
UsbCapture::run(): Submitting the transfers
UsbCapture::run(): 16 simultaneous transfers launched.
PlayerControl::stopAutomaticCapture(): Automatic capture not running, ignored.
UsbCapture::run(): Transfer stopping - waiting for in-flight transfers to complete...
UsbCapture::runDiskBuffers(): Thread stopped
`
<ten second pause>
`
UsbCapture::run(): Transfer stopping - Freeing transfer buffers...
UsbCapture::run(): Transfer stopping - waiting for disk buffer processing to complete...
UsbCapture::freeDiskBuffers(): Freeing disk buffer memory
UsbCapture::run(): Transfer complete, 3102 transfers performed with 11 disk buffers written
`

simoninns, 2018-09-24:
Can you confirm that you see "UsbCapture::runDiskBuffers(): Thread stopped" before the pause?  Seems very strange that the code would pause after all the threads have aborted (hence the double-check).

wondras, 2018-09-24:
Tested again; that is definitely the last message before the pause.

simoninns, 2018-09-24:
Ok, thanks; that makes it a lot more interesting :)  I'll take a look and see if I can spot what could be causing it.

simoninns, 2018-09-24:
I added a possible fix in commit 2f0c84d - the change is now merged in the master branch.  Please retest.

simoninns, 2018-09-24:
Also - downgrading from bug to enhancement - since the capture functionality is working (and Mac OS support is not project-core)

simoninns, 2018-09-24:
Fix in latest commit did not change the issue.  Needs some investigation by someone with a Mac to test on :)

wondras, 2018-09-24:
DdD calls libusb_handle_events_timeout() while waiting for in-flight transfers to complete. The docs say this function is deprecated due to possible race conditions:

http://libusb.sourceforge.net/api-1.0/group__poll.html#ga6deff4c7d3a6c04bb9ec9fd259b48933

It recommends using different functions that handle a completion flag more directly. I wasn't quite clear on how to implement them in this context to do a test, but this may be worth a look.

simoninns, 2018-09-28:
This issue may have been helped by commit 843163d.  Please retest if possible.

wondras, 2018-10-07:
Confirmed working in current build; no more delay.

closed 2018-10-07 by wondras