Domesday86/DomesdayDuplicator #53: Timer does not stop when capture is halted due to disk buffer overflow

wondras, 2018-09-26:
Test case:
1. Start a capture
2. Do stuff on the computer that causes disk writes to slow down, such as copying other files on the same drive

The capture will halt due to buffer overflow. Messages appear in the console, but nothing appears in the GUI. The timer continues to count up, though the file size does not. 

Clicking to stop transfer has no effect, and clicking it again causes the app to stop responding.

Suggested:
Add some sort of notification/pop-up that indicates capture has failed, and reset to idle state.

simoninns, 2018-09-26:
I ran a test case for this on Ubuntu (which I'm assuming is your environment too).  A disk buffer full condition results in the following:

[image]

...and the debug contains:

UsbCapture::run(): Submitting the transfers
UsbCapture::run(): 16 simultaneous transfers launched.
UsbCapture::runDiskBuffers(): Thread started
bulkTransferCallback(): Disk buffer overflow error!
UsbCapture::run(): Transfer stopping - waiting for in-flight transfers to complete...
UsbCapture::runDiskBuffers(): Thread stopped
bulkTransferCallback(): Disk buffer overflow error!
bulkTransferCallback(): Disk buffer overflow error!

Can you confirm the output you are seeing from the console debug (cut and paste some of it here)?

wondras, 2018-09-26:
On startup:
```
Configuration::readConfiguration(): Reading configuration from "/home/warren/.config/DomesdayDuplicator.ini"
MainWindow::startPlayerControl(): Player type is not configured in preferences
PlayerControl::configurePlayerCommunication(): No player configured, control thread stopped
UsbDevice::scanForDevice(): Scanning for the USB device...
MainWindow::deviceAttachedSignalHandler(): Domesday Duplicator USB device has been attached
UsbDevice::run(): libUSB event poll thread started (using hot-plug events)
'''

On capture start:
```
MainWindow::on_capturePushButton_clicked(): Starting capture to file: "/media/warren/4TBtoGo/rf/RF-Sample_2018-09-26_17-40-05.lds"
MainWindow::on_capturePushButton_clicked(): Starting transfer
UsbDevice::startCapture(): Starting capture
UsbDevice::startCapture(): Sending start capture USB vendor specific command
UsbDevice::startCapture(): Opening the capture device
UsbDevice::startCapture(): Creating the capture object
UsbDevice::startCapture(): Starting capture process with start()
MainWindow::on_capturePushButton_clicked(): Transfer started
UsbCapture::run(): Setting up the transfers
UsbCapture::allocateDiskBuffers(): Allocating memory for disk buffers
UsbCapture::run(): Submitting the transfers
UsbCapture::runDiskBuffers(): Thread started
UsbCapture::run(): 16 simultaneous transfers launched.
```

On overflow:
```
bulkTransferCallback(): Disk buffer overflow error!
UsbCapture::run(): Transfer stopping - waiting for in-flight transfers to complete...
bulkTransferCallback(): Disk buffer overflow error!
bulkTransferCallback(): Disk buffer overflow error!
bulkTransferCallback(): Disk buffer overflow error!
bulkTransferCallback(): Disk buffer overflow error!
bulkTransferCallback(): Disk buffer overflow error!
bulkTransferCallback(): Disk buffer overflow error!
bulkTransferCallback(): Disk buffer overflow error!
bulkTransferCallback(): Disk buffer overflow error!
bulkTransferCallback(): Disk buffer overflow error!
bulkTransferCallback(): Disk buffer overflow error!
bulkTransferCallback(): Disk buffer overflow error!
bulkTransferCallback(): Disk buffer overflow error!
bulkTransferCallback(): Disk buffer overflow error!
bulkTransferCallback(): Disk buffer overflow error!
bulkTransferCallback(): Disk buffer overflow error!
UsbCapture::run(): Transfer stopping - Freeing transfer buffers...
UsbCapture::run(): Transfer stopping - waiting for disk buffer processing to complete...
UsbCapture::runDiskBuffers(): Thread stopped
```

On clicking start/stop button to stop:
```
PlayerControl::stopAutomaticCapture(): Automatic capture not running, ignored.
```
After this click, the program freezes and has to be killed.

simoninns, 2018-10-01:
Looking at the debug, I can't see the line "UsbCapture::run(): Transfer failed - emitting notification signal" in the traces; so it could be that the usbCapture::run() method isn't seeing the transferFailure flag (it's becoming corrupt somehow).  So I've moved it to a static volatile to make it more 'callback safe' in commit 6179203

Please retest.

simoninns, 2018-10-02:
As this looks to be fixed, I'm closing the issue.  Please reopen if the problem reoccurs.

closed 2018-10-02 by simoninns