Domesday86/DomesdayDuplicator #72: USB device fails to attach on start-up

simoninns, 2018-11-26:
When starting the USB application (after a host reboot) the USB device doesn't seem to attach correctly:

Debug: [../DomesdayDuplicator/usbdevice.cpp:217] UsbDevice::scanForDevice(): Scanning for the USB device...
Debug: [../DomesdayDuplicator/usbdevice.cpp:170] UsbDevice::run(): libUSB event poll thread started (using hot-plug events)
Debug: [../DomesdayDuplicator/usbdevice.cpp:292] UsbDevice::open(): Found device with matching VID/PID, but attempting to open it failed!

simoninns, 2018-11-26:
I'm not sure what's causing this as the USB device debug looks good on start-up and the Linux udev rule seems to be applied (re-running the udev rules does not make the device attach correctly after boot).

This could be an issue related to the USB controller on my dev machine... if any one else sees a similar issue please report it here.

IanSmallshire, 2021-05-20:
I have managed to get a log from the boot of my laptop.

Laptop is dual boot Win10 and 20.04 with GRUB 4 windows. This causes the Ubuntu to load showing all of the status text and not the loading screen.

My Testing stages

- Clean boot from cold without ddd attached and then attach to the machine after boot OK
- Clean boot from cold with ddd attached and duplicator app reorts ddd not connected - detatch and reattach and OK.
- Reboot after good connection to ddd has successfully initialized causes different state during startup and adds 50+ seconds to the startup time.

I recorded a video of boot and here is a screen capture of the status
[image]

simoninns, 2026-08-20:
Applies to the legacy firmware

closed 2026-08-20 by simoninns