Domesday86/DomesdayDuplicator #84: Dual channel ADC for VHS tape capture

simoninns, 2019-06-17:
For use with VHS tape systems (and other capture scenarios where there are dual sources of analogue data), it would be useful for the duplicator board to provide 2 synchronised channels of ADC running from the same sample clock.

This would need some testing though as it's not clear if the USB3 can sustain 2 40MSPS streams (testing so far has been up to 75MSPS).

happycube, 2019-06-17:
VHS doesn't need anywhere near 40msps to work effectively, so there's definitely enough bandwidth there.

simoninns, 2019-06-17:
Good point - since the ADC filter is fixed, the FPGA could be used to decimate the sample in software (to lower the sample rate without changing the ADC clock) - which would then keep the design more universal.

harrypm, 2023-08-28:
Just thought I would add a footnote 

- 16msps VHS NTSC 
- 18msps VHS PAL 
- 20msps Is considered safe for all VHS
- 24msps Is considered safe for all SVHS
- 8msps Is considered safe for HiFi (but less can work depending on format)

While mimium sampling is great for post space saving oversamping for VHS is plus in the event of dropouts or out of spec signals.

Video8 / Hi8 fall into 20/24msps safe relm and are 1 RF tap capture formats.

harrypm, 2023-12-17:
Just going to note the new [adc_usb3 / SMIRF](https://github.com/oyvindln/vhs-decode/wiki/RF-Capture-Hardware#adc_usb3--smirf), its in a very dev board work in progress state and software/audio boards are a work in progress aswell, but it has great potential.

Cost: 150 Euro +- 25 Euro - JLCPCB/PCBway direct fabrication

PCB: 20-30USD / Parts 50-70USD

Features:

- Offshelf FX3 USB 3.0 board (same as DdD)
- External Clock Source Output
- 6 Extra Aux inputs for audio ADC modules etc
- Duel ADC / Duel Imput
- Physically adjustable input filters
- 12-Bit, 20/40/65msps sampling
- AD8138 Driver, Op-Amp / SN74ls541 Buffer / AD9235 ADC