Domesday86/DomesdayDuplicator #85: Direct composite signal capture

simoninns, 2019-06-17:
It would be useful to be able to use the Duplicator board for direct 1V PtP composite video capture.  Requires the ability to widen the input voltage range and (most likely) vary the sampling rate from the GUI

atsampson, 2022-05-17:
You'd also need to make C401 significantly bigger (47µF?). The second image here is what capturing composite video with the DDD looks like at the moment:

[image]

harrypm, 2022-12-18:
I recently modified my DdD for CVBS testing and dumped all my sample data on the public drive for everyone to play with.

I'll be testing 22uF ~ 47uF in the future when I have more of a stock of SMD 0805 parts handy.

[Discord Link](https://discordapp.com/channels/665557267189334046/824771223270981672/1048689615814541392) to the related conversation - December 2022

[C401 3.3uF Samples](https://drive.google.com/open?id=1rerpI89oVtHFr4gmfjAQYw8TGMWV0PIa) 

More [CVBS RAW Sample Data]()

[image]

It required a 50Ohm load resistor or the input level was too high on minimum gain.

[image]