Domesday86/DomesdayDuplicator #88: Replace J401 (BNC connector)

simoninns, 2019-06-17:
The BNC connector J401 (BNC Socket – Rosenberger 51K204-400A5) is a non-generic and relatively expensive choice.  Could be replaced by a BNC connector that is more generic and has a standard foot-print.

DreckSoft, 2020-07-24:
That would actually be very nice as the connector is the second most expensive part which does not make sense for such a common part.

For example this one:
https://www.aliexpress.com/item/33003384739.html
has a similar layout. Unfortunately it does not fit as a 1:1 replacement as the outer pins are only 8mm apart. I'm currently try to figure out how to change this in KiCAD. Also the connector should be moved more to the edge of the board as this one is shorter than the Rosenberger. Or just soldier it 90 degrees rotated, that also works.

simoninns, 2020-07-24:
You have to be a little careful here as the RF path before the pre-amp stage (the high-bandwidth opamp) is very susceptible to noise; the choice of connector and opamp were based on quality rather than price.  Also generic parts from suppliers with little to no quality assurance (such as aliexpress) can cause knock on support issues since there is no way to ensure the quality from purchase to purchase.

Also note that the chosen connector much match the impedance of the board's input (50 ohms).

DreckSoft, 2020-08-05:
Ok I see. I tried a replacement connector using the original center hole and just cutting the ground pins and soldered the housing to the original ground connections. There was NO difference in SNR at ~ 1/15th of the cost. 

Maybe we could add matching ground connections as an alternative? That shouldn't have any impact, cause the signal path isn't changed. The Rosenberger seems to be 11mm wide, the replacement only 9mm. Meaning the connectors would be very close. I'm not sure if it's possible to do a "double hole" or even a line that would support all center pin connectors within a certain range in the design. Just a suggestion, I can perfectly live with the current layout. I added pictures for both connectors and an example how it could look.

Of course the issue with the quality assurance remains. I can just speak for my batch. 

[image]
[image]
[image]
[image]