Domesday86/DomesdayDuplicator #96: Supported LaserDisc hardware?

iancward, 2019-09-21:
The README indicates that this will work on a Philips VP415 but the [user guide page](https://www.domesday86.com/?page_id=1312) seems to indicate that the  Pioneer LD-V4300D is also supported. What about LD-V4400 or the LD-V2000?

simoninns, 2019-09-21:
The duplicator will work with pretty much any LaserDisc player - the prerequisite is access to a service manual for the player in order to locate the RF output test-point (where the duplicator is attached).

The reference player (i.e. the hardware against which the duplicator is tested) is the LD-V4300D - that is simply because I don't have the cash to buy hundreds of players and test against all of them.

If you want to capture many discs it's also recommended to get a player that supports serial control - the Duplicator GUI supports many Pioneer players and can automatically control and monitor the capture process.

If you want to capture for preservation, or simply to get the best results, the player should be calibrated too (this procedure is covered in the service manuals).

iancward, 2019-09-21:
Ah, ok, maybe I should have read a bit more in the repository. I had been thinking this was something that would connect to the serial port on the back of some devices. Or do you use the serial port for control and the test port for capture?

My main intent is for archival purposes, trying to find a better option than setting up a capture device. But at this point, I'm starting from scratch, so I'm trying to determine what player and equipment to start with.

simoninns, 2019-09-22:
There is a project facebook group you can join if you want to discuss various options (Domesday86) and we also have a joint IRC channel on #freenode where you will find more technical/development discussion #domesday86

closed 2019-09-22 by simoninns