Domesday86/DomesdayDuplicator #104: User Guide changes (https://www.domesday86.com/?page_id=1312)

ziggy90127, 2020-04-17:
https://www.domesday86.com/?page_id=1312

"sudo apt install git g++ qtbase5-dev qt5-default qt5.qmake libqt5serialport5-dev libusb-1.0"
->
"sudo apt install git g++ qtbase5-dev qt5-default qt5.qmake libqt5serialport5-dev libusb-1.0 make"

"Linux-application" -> "Linux-Application" 4 instances.

DreckSoft, 2020-08-05:
There are a few more things I notices during the setup.

echo "-1" > /sys/module/usbcore/parameters/autosuspend
This disables USB autosuspend and is an MUST to Flash the DE0 at least on my Dell Laptop. Otherwise the flash fails. The capture later on will also be useless if it's not disabled.

https://www.domesday86.com/?page_id=1312
sudo apt install git g++ qtbase5-dev qt5-default qt5.qmake libqt5serialport5-dev libusb-1.0
It should be "qt5-make", so - instead of .

Additionally, you will at least need 
libusb-1.0-0-dev
libqt5-dev
and some qt5 components which I also don't remember.

The .cov expects a certain path.
Just edit the .cof file and remove the path. Quartus will search for it in the same directory, so it'll be universal.

You might want to link this repo:
https://github.com/Ho-Ro/cyusb_linux
That compiles cleanly with QT5 on Ubuntu 20.04

After compilation you need to re-link the lib, as /usr/local/lib is no longer in PATH
ln -s /usr/local/lib/libcyusb.so /usr/lib/libcyusb.so

lizardgai4, 2022-08-11:
I suggest these changes to this webpage:

https://www.domesday86.com/?page_id=1312#Getting_the_software

`sudo apt install git g++ qtbase5-dev qt5-default qt5-qmake libqt5serialport5-dev libusb-1.0`

Remove "qt5-default"
Replace "libusb-1.0" with "libusb-1.0-0-dev"

Changed version:

`sudo apt install git g++ qtbase5-dev qt5-qmake libqt5serialport5-dev libusb-1.0-0-dev`

Also, I opened another issue regarding the readme and installation.  It currently directs to https://www.domesday86.com/?page_id=978, but should direct to page_id=1312

simoninns, 2022-08-29:
Note, page 1312 was the User Guide which has now moved to:

https://github.com/simoninns/DomesdayDuplicator/wiki/User-Guide

I haven't fixed anything yet, but just wanted to note the URL mapping.

simoninns, 2022-12-19:
Closed, fixed.

closed 2022-12-19 by simoninns