Domesday86/DomesdayDuplicator #123: DdD Application: Side-specfic mint mark/notes

Gamnn, 2022-08-29:
For two-sided discs in players that support two-sided playback, it would be easier to input the mint marks if the field was tied to the side number, so both can be registered before beginning capture.

In other words, you enter the mint marks on side 1, increase it to side 2 and enter side2's marks. When you decrease back to side1, the previous input for side1 would be re-populated in the mint marks field. 

Ideally, when increasing or decreasing the side selection, it would keep the last entered input in the mint marks field until manually written over, rather than starting with a blank field.

The same could be implemented for the notes field.

TokugawaHeavyIndustries, 2022-08-29:
I'll take care of this. I'll get it implemented soon

TokugawaHeavyIndustries, 2022-08-30:
Just opened PR for this, #125

TokugawaHeavyIndustries, 2022-09-01:
There we go, #127

simoninns, 2022-12-17:
PR accepted; issue closed

closed 2022-12-17 by simoninns