Domesday86/DomesdayDuplicator #129: Build instructions broken on Ubuntu 22.04 LTS

TokugawaHeavyIndustries, 2022-09-09:
The build instructions for the capture application are broken for Ubuntu 22.04 LTS.

For QMake:  QT5 is no longer available in normal repos for 22.04

For CMake:  The following error is outputted:
`CMake Error at CMakeLists.txt:16 (qt_add_resources):
  Unknown CMake command "qt_add_resources".`

atsampson, 2022-09-09:
For Qt 6, it builds OK with Ubuntu 22.04 and the suggested list of dependencies [in the CI](https://github.com/simoninns/DomesdayDuplicator/blob/master/.github/workflows/tests.yml#L40). The CMake files for Qt are in `qt6-base-dev`, so if the dependency installation succeeded then you should have them. Can you show the full output from running the commands in the instructions, and check whether there's anything odd in your environment (`env`)?

(Qt 5 is still available in 22.04 as well, but the qt5-default package doesn't exist any more.)

lizardgai4, 2022-09-09:
This is strikingly similar to issue #104

simoninns, 2022-12-19:
I believe this is fixed now; so closing

closed 2022-12-19 by simoninns