Domesday86/DomesdayDuplicator #132: LD-V4300 Calibration

hilios5972, 2022-11-06:
Hi, I didn't understand if you need two remotes or just one remote, thanks.

TokugawaHeavyIndustries, 2022-11-06:
Hello @hilios5972 , it's stated under "Required Materials" the following:

Pioneer CU-V113A LaserDisc IR remote control
RM Series RMC12010 3rd-party Pioneer service IR remote

These are two different remotes (note the model numbers)

hilios5972, 2022-11-06:
ok they are different and I understand, but do they serve together?  Inviato da Posta per Windows 10 Da: Matt PerryInviato: domenica 6 novembre 2022 19:30A: simoninns/DomesdayDuplicatorCc: hilios5972; MentionOggetto: Re: [simoninns/DomesdayDuplicator] LD-V4300 Calibration (Issue #132) Hello @hilios5972 , it's stated under "Required Materials" the following:Pioneer CU-V113A LaserDisc IR remote controlRM Series RMC12010 3rd-party Pioneer service IR remoteThese are two different remotes (note the model numbers)

TokugawaHeavyIndustries, 2022-11-06:
They are both required.  An example of this is "Grating temporary adjustment"

hilios5972, 2022-11-06:
ok thanks I will try to find them for sale but it will be difficult. Inviato da Posta per Windows 10 Da: Matt PerryInviato: domenica 6 novembre 2022 19:44A: simoninns/DomesdayDuplicatorCc: hilios5972; MentionOggetto: Re: [simoninns/DomesdayDuplicator] LD-V4300 Calibration (Issue #132) They are both required. An example of this is "Grating temporary adjustment"

wondras, 2022-11-06:
I don't know if the codes for the RMC12010 are readily available, but I've been able to get by for some newer players using a universal remote app and a phone with a headphone jack. You can buy a little IR 'nub' on eBay for a few dollars, or even build one yourself with a couple of IR LEDs wired with opposing polarities. Each LED flashes on one half of the audio waveform cycle, so a 19 KHz tone creates the 38 KHz IR carrier. (A very clever hack, I must say!)

At the time, I used a somewhat dodgy-looking app called ZaZa Remote, which I believe is still available for iOS and Android. 

If we could establish a known-good set of codes for a known-good app, it might save a lot of time spent tracking down obscure remotes (and driving up the prices of them in the process.)

atsampson, 2022-11-06:
You don't need two remotes, and you may already have one that'll do the job if you have other Laserdisc players.

I normally use the remote from my CLD-1450 - that has all the buttons you need to calibrate an LD-V4300D/LD-V4400, since you can get those players into service mode by holding DISPLAY on powerup. It doesn't have the ESC button that you need to get consumer players into service mode, but many of them have an alternative way of doing that described in the manual (typically grounding one of the CPU lines on powerup).

If you don't have a real remote at all, you can use a Linux machine running LIRC, e.g. a Raspberry Pi - [here's my config file](http://offog.org/git/misccode/lirc/PioneerCLD.conf) with a superset of various Pioneer remotes.

hilios5972, 2022-11-06:
Thanks, i have a cld-1600 do you think it is okay?   Da: Adam SampsonInviato: domenica 6 novembre 2022 21:54A: simoninns/DomesdayDuplicatorCc: hilios5972; MentionOggetto: Re: [simoninns/DomesdayDuplicator] LD-V4300 Calibration (Issue #132) You don't need two remotes, and you may already have one that'll do the job if you have other Laserdisc players.I normally use the remote from my CLD-1450 - that has all the buttons you need to calibrate an LD-V4300D/LD-V4400, since you can get those players into service mode by holding DISPLAY on powerup. It doesn't have the ESC button that you need to get consumer players into service mode, but many of them have an alternative way of doing that described in the manual (typically grounding one of the CPU lines on powerup).If you don't have a real remote at all, you can use a Linux machine running LIRC, e.g. a Raspberry Pi - here's my config file with a superset of various Pioneer remotes.

hilios5972, 2022-11-06:
Now I am looking for the test disc, I have found a GGV1016 you think it works, the ggv1069 I can not find it.  Inviato da Posta per Windows 10 Da: Adam SampsonInviato: domenica 6 novembre 2022 21:54A: simoninns/DomesdayDuplicatorCc: hilios5972; MentionOggetto: Re: [simoninns/DomesdayDuplicator] LD-V4300 Calibration (Issue #132) You don't need two remotes, and you may already have one that'll do the job if you have other Laserdisc players.I normally use the remote from my CLD-1450 - that has all the buttons you need to calibrate an LD-V4300D/LD-V4400, since you can get those players into service mode by holding DISPLAY on powerup. It doesn't have the ESC button that you need to get consumer players into service mode, but many of them have an alternative way of doing that described in the manual (typically grounding one of the CPU lines on powerup).If you don't have a real remote at all, you can use a Linux machine running LIRC, e.g. a Raspberry Pi - here's my config file with a superset of various Pioneer remotes.

simoninns, 2022-12-19:
Closing as non-issue.  If documentation is unclear around need for remotes; please open an issue with suggested changes.

closed 2022-12-19 by simoninns