Domesday86/DomesdayDuplicator #136: Feature request: Auto-stop on low amplitude threshold (after a configurable delay)

JonasCz, 2022-12-28:
I'm a complete C++ noob, so I think have no chance of implementing this well myself, but just in case someone would like to take this on:

I'd love to see an automatic stop feature, triggered when the amplitude measurement drops below a certain threshold for a configurable amount of time.

My usecase is VHS captures, for which the amplitude drops to a very low level when the on-tape recording ends. This would help to  avoid creating needlessly big files.

Thanks for creating this amazing project!

JonasCz, 2022-12-28:
Thanks, that works really well! And it's fast too.

I didn't know that it's possible to truncate ("shorten") a file in-place without a copy operation, learn something new every day :-)

simoninns, 2026-08-18:
duplicate of #82

closed 2026-08-18 by simoninns