Domesday86/DomesdayDuplicator #143: [not really issue] ADV7181D

Randrianasulu, 2023-02-11:
Hello, and thanks for gian amount of work put in such nuche project.

We at cinelerra-gg mail list were looking at vhs (but also hdmi) capture, and it seems most capture cards/boxes only do at best 8bit yuv 4:2:2 . Not bad if you get this as-is ... but apparently some Analog Devices chips can do better?

May be I completely misread https://web.archive.org/web/20221006213255/https://www.analog.com/media/en/technical-documentation/data-sheets/adv7181d.pdf  but it seems to support not just 10-bit per component yuv, but up to 20 bits per component?! (non-standart).

I wonder if fpga-usb3 parts of your project can be reused for interfacing with this IC? Yes, it  just black box IC , quite contrary to project's main goals of being max open. But I found whole v4l2 subsystem (not used here, from what I see) lack support for formats like v210 , so may be existence of linux device actually  capable of outputting such signal will stimulate some development there, too?

I found it interesting how 10-bit (per component) digital video was all the rage in year 1994 or so (on big SGI machines).

As asude there  was "rgb conditioner" fpga project around, at much slower speed, but still trying to deal with unstable input signal, they also manage to use fmpa itself instead of ADC!

https://github.com/svofski/videoconditioner

DC-HDMI also looked interesting (Dreamcast output to hdmi devices), but as far as I understand it reads already digital signals, not converting analog from usual outputs.

harrypm, 2023-02-11:
@Randrianasulu I should probably guide you to the [community discord](https://discord.gg/tRFYK5NNnE) at this point as that's where all the projects devs and testers for capture/decoding and a lot of people from various A/V fields are in there.

But I will also note here the [ADV8472](https://www.analog.com/media/en/technical-documentation/data-sheets/adv7842.pdf) is pretty much the end of the analogue all-in-one processing IC's if you give it enough ram for its TBC/Framestore its used in almost everything off the shelf from [Blackmagic's poorly regarded SDI units](https://www.blackmagicdesign.com/products/miniconverters) to Magewells top end [PCIe desktop AIO](https://www.magewell.com/products/pro-capture-aio) and [USB 3.0 AIO](https://www.magewell.com/products/usb-capture-aio) based units use the same chip and Xilinx Spartan 7 FPGAs.

Ware as Quadregia uses ADC's then FFPGAs for real-time but this is only for the Betacam formats as far as I know they just use the ADV stuff for VHS etc.

Randrianasulu, 2023-02-11:
@harrypm , yeah, thanks for invite (while I mostly hang around instead of doing stuff).

I hope some extracts from discord sessions stull periodically edited back into main documentation ( I've heard Discord  not very user-friendly when it comes to searching and/or using non-official clients. I use web-version, so I guess it works for me ...)

harrypm, 2023-02-11:
@Randrianasulu I filter current info and keep on top of current docs with whatever goes on inside the discord, but yeah you can do quite a targeted lookup of the entire multi-year history and or [dump the whole server to local text data](https://dht.chylex.com/) if you so wish don't be shy to ask things in there though!

Randrianasulu, 2023-02-11:
ok.

https://manualzz.com/doc/6859182/adv7181d---element14

seems to be programming manual, as opposed to just product brief. Still more ADVanced chip you suggested us bga-256, so you need a lot of balls (pun intended) to make good use of it. But at least now we know why high-end devices tend to have so much onboard memory

Randrianasulu, 2023-02-12:
so, at least register maos for adv7842 still open:

https://web.archive.org/web/20221006194914/https://www.analog.com/media/en/technical-documentation/user-guides/UG-206.pdf

this is good, but 128/256 Mb of ram also must come from somewhere, so cost++ , PCB complexity++ ... (sorry for stating obvious, I just liked whole 'lego' aspect of design, so I thought half of work can be reused by competent party to do something slightly different).

harrypm, 2023-02-12:
@Randrianasulu DDR2 SDRAM is nothing cost wise, complexity is why auto routers are a thing too, BGA chips can be [made into modules](https://www.advanced.com/pdf/AIC_BGADataBook_rev17-2.pdf) as well there is boards already out there for that you can always get a Blackmagic analogue to SDI units as cheep parts boards and put your own firmware one them too they are about 80-120USD a unit on the used market even less if there is a dead port on the SDI outputs considering the ADV chip its self is about 40USD MSRP from PCBways distributors last I checked its not bad cost for entry if you want to tinker.

Randrianasulu, 2023-02-12:
I personally can't even open my malfunctioning glued-in huawei tablet .. :( But idea about using parts of high-end boxes actually new for me, thanks!

harrypm, 2023-02-12:
@Randrianasulu Oh Blackmagic gear is just 4 philps screws then 4 more inside basically 2 pieces of stamped metal hence why these are cheep and die from overheating there is no gaskets or fans anything.

Randrianasulu, 2023-02-12:
also yes, idea about using evalution board as erzats captire device also visited me ...

but ...

https://eu.mouser.com/ProductDetail/Analog-Devices/EVAL-ADV7181DEBZ

A) it cost more than high-end box (after I looked for d revision of chip especially, not earlier one)

b) I have no idea how it *output* video. Hdmi and usb connectors exist, yet they can be inputs-only/programming ...

Also, from dept. of WRONG ideas - I thought if one fx2 is not enough for one >8 bit yuv422 SD  stream you can use two of them going to different usb 2.0 ports on receiving side ....

Randrianasulu, 2023-02-12:
aaah, hdmi-out with some IC

https://web.archive.org/web/20230212024225/https://ez.analog.com/cfs-file/__key/communityserver-wikis-components-files/00-00-00-01-10/2210.ADV7181D_5F00_Evaluation_5F00_Board_5F00_Documents.zip

Also default software seems to be windows-based :)

I am afraid something like http://www.triplespark.net/elec/periph/USB-FX2/ is more in my league ..and even then - not tested, by me!

Randrianasulu, 2023-02-12:
also from my backlog:

https://electronics.stackexchange.com/questions/311491/adv7180-breakout-board-critique-sought

Randrianasulu, 2023-02-12:
ADV7842-EVAL thread on videohelp gorum (2019-2023)

https://forum.videohelp.com/threads/394878-450-VHS-Hi8-10-bit-422-with-Frame-Line-TBC-Capture-Setup

requires another hdmi-pc interface, tho ...

I also wondered about how yuv traveled in hdmi ...seems to be  as 16bit!

https://www.liftgammagain.com/forum/index.php?threads/question-about-10-bit-panels-and-video-output-card.6581/page-2

(it might sound strange to worry about yuv-rgb conversion inside IC, but ...)

Randrianasulu, 2023-02-12:
somewhat offtop, but I also looked at aja cards (with Linux in mind) and surely there was driver supporting timecode:

https://github.com/aja-video/ntv2-v4l2
but for pci(e?) cards, not usb.

Up one level you can find another type of driver ... not v4l2

Randrianasulu, 2023-02-12:
ah, reddit thread, it seems more complex adv7842  more strictly controlled due to hdmi stuff...

https://www.reddit.com/r/LaserDisc/comments/m69l1l/evaluation_board_for_comb_filtering_and_tbc/

Randrianasulu, 2023-02-12:
Also from my email archive:
"Homebuilt digital video TBC/standards converter"

https://web.archive.org/web/20181219122351/http://marcusgun.awardspace.com/newvidproc.html

simoninns, 2023-08-20:
Closed as off-topic.  Please use the discord server for general discussions.

closed 2023-08-20 by simoninns