Domesday86/DomesdayDuplicator #146: 20 mhz mode

oyvindln, 2023-05-03:
This would be very useful for formats that don't need the full sample rate to save disk space and reduce risk of sample drops due to slow writes.

Not sure if this could be sorted simply in teh capture app with simple 2:1 decimation or if it requires extra filtering to avoid aliasing

wondras, 2023-09-09:
Here is a quick Python script you can use to see if simple decimation is viable:

`
#!/usr/bin/env python3

import sys
import numpy as np

BUF_SIZE = 1024*1024*10

while True:
    in_buf = sys.stdin.buffer.read(BUF_SIZE)
    if len(in_buf) == 0:
        break
    in_array = np.copy(np.frombuffer(in_buf, dtype=np.int16))
    out_array = in_array[::2]
    sys.stdout.buffer.write(out_array.tobytes())
`

It uses 16-bit raw input and output, and drops every other pair of bytes.

Simple test to show functionality:
`
$ echo 'This is a test 1234567890' | python3 ./decimation_test.py
Th ia st23670
`

Pipeline to decimate an LDS file:
`
$ ld-lds-converter -u -i test.lds | python3 decimation_test.py | ld-lds-converter -p -o test_decimated.lds
$ ls -l
total 29312
-rwxr-xr-x@ 1 warren  staff       317 Sep  9 11:42 decimation_test.py
-rw-r--r--  1 warren  staff  10000000 Sep  9 11:58 test.lds
-rw-r--r--  1 warren  staff   5000000 Sep  9 11:58 test_decimated.lds
`
(10 MB decimated to 5 MB)

simoninns, 2026-08-18:
Closing due to PR #172

closed 2026-08-18 by simoninns