Domesday86/DomesdayDuplicator #147: Will a DD Compatible with an HLD-X0?

KT7000, 2023-05-12:
I own an HLD-X0, but I'm not sure if a DD will be compatible with it since the one LD player I usually see people using for this is an LD-V4300D.

Does the X0 have an EFM port? If so, where?

TokugawaHeavyIndustries, 2023-05-12:
Any player that has an RF test point available inside the player is compatible.  There have been reports of MUSE players working.

The DDD does not attach to an EFM port.

KT7000, 2023-06-04:
> Any player that has an RF test point available inside the player is compatible. There have been reports of MUSE players working.
> 
> The DDD does not attach to an EFM port.

Will these links for the V4300D still help attempt it with a Muse player? 

[Overview](https://github.com/simoninns/DomesdayDuplicator/wiki/LDV4300D-Overview)
[Cleaning](https://github.com/simoninns/DomesdayDuplicator/wiki/LDV4300D-Cleaning)
[Adding an RF Output](https://github.com/simoninns/DomesdayDuplicator/wiki/LDV4300D-RF-Output)
[Calubration](https://github.com/simoninns/DomesdayDuplicator/wiki/LDV4300D-Calibration)
[PSU Recap](https://github.com/simoninns/DomesdayDuplicator/wiki/LDV4300D-PSU-Recap)
[GGV1069](https://github.com/simoninns/DomesdayDuplicator/wiki/GGV1069-Description)
[RMC1012010 Remote](https://github.com/simoninns/DomesdayDuplicator/wiki/RMC12010-Remote)

simoninns, 2023-08-20:
All information on the subject is useful - but you will need to study and understand the service manual for your player or find someone on the project discord service with specific experience of your player.  If you are not confident about attempting this (given the information available) I would suggest that you don't as you will risk damaging your player.

closed 2023-08-20 by simoninns