Domesday86/DomesdayDuplicator #149: Just to simplify this...

effiebaby, 2023-09-24:
Could we not use an oscilloscope and a logic analyzer connected to a computer to do the same thing here?

harrypm, 2023-09-24:
Does it cost less then 300USD?

Can it do 40msps 10-bit over USB 3.0 constant stream? 

The reality is no, the ossiliscopes that reach the spec are far more cost wise, there is [off shelf SDR units](https://github.com/oyvindln/vhs-decode/wiki/RF-Capture-Hardware)

There is also the [CX Cards](https://github.com/oyvindln/vhs-decode/wiki/CX-Cards) that can do 50msps 8-bit for less then 30USD. 

The DdD was perpouse mode for LD input filter wise (though I use it for VHS etc)

effiebaby, 2023-09-29:
What about the u3pro16?  Not trying to be rude just understand.

harrypm, 2023-09-29:
@effiebaby It's not continuous stream it's a 16GB buffer from what I am reading.

Handy for getting test samples, potentially could work for hi-fi RF, but not actually useful for full capture runs, it has to be continuous, as in capable of going for hours.

We have a working list of candidates for off shelf ADCs [here](https://github.com/oyvindln/vhs-decode/wiki/RF-Capture-Hardware) on the VHS-Decode (does more then VHS..) wiki which has the majority of current documents.

99% of osiliscopes are buffered memory or far too low continuous stream, SDR units have broken the barrer, but required extra hardware to get the filtering range correct, and unlike the the DdD do not have plug and play support for Windows/MacOS/Linux like the [DdD does](https://github.com/harrypm/DomesdayDuplicator#windows-software-installation).

The nearest thing that could be a genuine off shelf successor is the thunder scope however it costs far more 500USD+ range, a modified RX-888 MKII, could potentially be a solution also, but the software support is just not there yet.

There is a whole community for the FM RF archival of many formats gathered on the [Discord Server](https://discord.gg/pVVrrxd) several people myself included who are trying to find more hardware solutions, it takes knowing what you're looking at and having money to trigger pull and experiment.

effiebaby, 2023-09-29:
They have a buffer and a stream mode.  If I am correct.  I thought the
stream mode can be directly sent through usb to be saved.  But as you said
software may not be there.   I appreciate the information.

closed 2026-08-18 by simoninns