Domesday86/DomesdayDuplicator #153: Prefer "Picture Stop" command over multi-speed forward for capable players.

dude22072, 2023-12-12:
According to the programmers manuals, the CLD-V2800 and CLD-V5000 both support the command `1PS` to disable stop codes (and `0PS` to enable them) to play through the entire disc uninterrupted. This should probably be preferred over the current multi-speed forward workaround on these players.

simoninns, 2026-08-18:
Documentation check — the `PS` command is confirmed in the primary manuals for both players.

## CLD-V2800

*Level III User's Manual*, TP 120 v. 1.0, 10/95. Command 53, listed in Appendix A under **"New LD Command"**, marked CAV-only:

> **Format:** `Integer PS`
>
> `0PS` (default): "Picture Stops are enabled and player stops at the picture stops encoded on a CAV disc until another motion command is sent to the player."
>
> `1PS`: "Picture Stops are disabled and the player plays the video material without holding a still frame at the picture stops."
>
> "Picture stops are encoded on CAV discs only, by request. Picture stops cannot be encoded on CLV discs."

Source: https://www.manualsdir.com/manuals/173359/pioneer-cld-v2800.html?page=65 (command list at `?page=25`)

## CLD-V5000

*Level III User's Manual*, TP 121 v. 1.0, 1/96. Same command, numbered 57, on manual page 3-49, with identical `0PS`/`1PS` semantics and the same `0PS` default.

Source: https://www.manualsdir.com/manuals/189961/pioneer-laserdisc_compact-disc-player-cld-v5000.html?page=65

## It really is V2800/V5000-only

Worth stating explicitly, since every Pioneer definition in `ddd-gui` currently inherits from the shared `LevelIII()` base in `src/player/players/pioneer_level_iii.h`:

- **LD-V4400** — the vendor manual already vendored in this repo (`docs-vendor/LD-V4400 Level I and III Documentation/`, Appendix A) lists all 50 Level III commands, and there is no `PS`. The manual only describes picture stops as something that happens to the player: "the player will play to that frame, stop and hold a still frame, until it receives the next motion command."
- **CLD-V2400 / CLD-V2600** — no `PS` in their Programmer's Reference either. Appendix J, note 2 simply says "If a picture stop code is recorded on currently playing CAV LaserDisc, the player enters Still Mode at the stop code point," with no override offered.
  (https://www.manualsdir.com/manuals/189268/pioneer-laserdisc_compact-disc-players-cld-v2400-laserdisc_compact-disc-players-cld-v2600.html?page=141)

The "New LD Command" grouping in the V2800 manual is the tell — `PS` was added for that generation. So this can't go into `LevelIII()`; it has to be a per-model override of `kPlayWithoutStopCodes` on `pioneer_cld_v2800.h` and `pioneer_cld_v5000.h` only, replacing the inherited `"PL64RBMF"` with something along the lines of `"1PSPL"`.

## Still needs bench verification

The manuals confirm the command exists and what it means, but not these, which all bear on how the override should be written. Neither player is available here for testing:

1. Is `1PS` accepted in Park (i.e. before `PL`), or only after spin-up? This decides whether `"1PSPL"` works as a single concatenated command the way `"PL64RBMF"` does.
2. Does the setting persist across `RJ` / tray cycles? If so, the capture sequence should probably re-send `0PS` to restore the documented default.
3. What is returned for `1PS` on a CLV disc? `auto_capture_sequence.cpp` already skips this path for CLV, so likely moot, but an `E04` would matter if it ever leaked through.
4. The `64RB` squelch component of the current workaround exists to keep audio on through multi-speed. With `PS` the disc plays at normal speed, so it should be unnecessary — worth confirming nothing else depended on it.

simoninns, 2026-08-18:
Since I don't have either of the players - this needs verification before it can be implemented (or a PR from someone who can test locally)