Domesday86/DomesdayDuplicator #156: How to choose the correct dip-switch gain setting?

darian-au, 2023-12-24:
I have recently discovered your fantastic project and decided to have a Duplicator fabricated using the PCB-Way turn-key project. After spending much time reading and learning how to build and flash the firmware I finally got it working. When I set all the dip-switches to ON (1111), ie the lowest gain setting and make a test capture, ld-decode is unable to detect any frames and continually skips until it reaches the end of file. When I set the gain to 8.5 (1000) ie the highest setting, ld-decode works and runs normally, giving me a result I can view in ld-analyse. 

There are 13 other gain settings I could choose from, how do I determine the most optimal setting for my laserdisc player? 

I understand the reasoning behind making the gain selectable as most laserdisc players are different and vary in their output RF levels. Should I use my multimeter or oscilloscope to measure the voltage or other value to then perform a calculation to choose the gain? I was unable to find any documentation about choosing the gain, other than the table specifying what the values actually are.

TokugawaHeavyIndustries, 2023-12-24:
Ideal inbound amplitude is around 0.75. Whatever gain setting though, you'll want zero clipping.

The latest version of the capture application has a real-time amplitude measurement and graph - you can use that for a general idea of the inbound amplitude.

A sure-fire way of getting it dialed in is capturing in 16-bit unsigned and opening the resultant capture in Audacity or another audio processing program (like Adobe Audition) and checking the amplitude settings there.

Remember each disc may require different settings for optimal capture, and in general, the amplitude increases as the track gets closer to the center of the disc.

simoninns, 2026-08-18:
Closing due to PR #172

closed 2026-08-18 by simoninns