Domesday86/DomesdayDuplicator #162: Error installing on Mac

jerauf, 2025-02-19:
### Checklist

- [X ] I have searched the issues page for any duplicate issues open or closed and confirmed that this bug has not been reported before.
- [ ] I have tested the issue with the current build.
- [ ] I have attached log files, uploaded sample data, and commands used so that the issue can be easily reproduced by the developers.

### Bug Description

Tried installing on Mac and got the following error:

```
By not providing "FindQt6.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt6", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt6" with any of
  the following names:

    Qt6Config.cmake
    qt6-config.cmake

  Add the installation prefix of "Qt6" to CMAKE_PREFIX_PATH or set "Qt6_DIR"
  to a directory containing one of the above files.  If "Qt6" provides a
  separate development package or SDK, be sure it has been installed.

```

### Steps to Reproduce

1. Followed Mac instructions https://github.com/harrypm/DomesdayDuplicator/wiki/Software-Setup-User-Guide#installing-the-software
2. Installed CMAKE
3. Ran make -j8
4. Above error

### Expected Behaviour

Installed DdD app.

### Actual Behaviour

Received above error.

### Environment

- Operating System: Sequoia 15.3.1
- Hardware Used: Mac Mini 2023 M2 Pro

harrypm, 2025-02-20:
https://github.com/simoninns/DomesdayDuplicator/issues/162

snoopy870, 2025-04-08:
Same issue here. Any fix?

simoninns, 2026-08-18:
Closing due to PR #172

closed 2026-08-18 by simoninns