Domesday86/DomesdayDuplicator #165: Request Simple Command Line Start Option

marshalleq, 2025-07-18:
Hi, I have searched through GitHub and seen that some refactoring has happened to support a future command line method of operation.  However I actually came here from the vhs decode project. 

I would like to raise awareness / request around the need for a simple command to start a capture with the DdD capture utility from the command line so that a simultaneous audio capture can be triggered for other projects using the DdD hardware.

 A number of us built domesday devices not realising that the audio isn’t simultaneously captured which we can work around with the clockgen lite project.  However this requires something like Sox to capture which ideally should be triggered simultaneously to avoid manual post alignment.

With the DdD being the superior capture device it makes more sense to do this than to go and build or buy something else for the purpose.  

The configuring of DdD settings can be done via the gui, as well as the stop capture step. Differing finish times do not really matter as these will have to be muxed together anyway.  All that we’d need is a command line method to start the capture which should also bring up the gui. How hard would that be?  Is this something that someone would be prepared to take on board as part of any future work?

Thanks.

simoninns, 2026-08-18:
It would be possible to start the GUI already in capture mode - but that would not give a deterministic start point (the GUI would load at it's own speed and then, some time after the command, capture would begin.  I'm not sure what is really needed as nothing will really provide 'sync'?  Would starting the GUI with capture running be enough?

marshalleq, 2026-08-18:
Yes the GUI is the problem in this scenario and that's the reason for the request for command line options to control the domesday duplicator.  On my setup as others have done, we use something like the clockgen lite https://github.com/namazso/cxadc-clockgen-mod which pairs beautifully with the Domesday Duplicator, syncs the clock between audio and video to prevent drift.  With the DdD command line addition, it's then possible to calculate the A/V delay / difference and get proper alignment.  The PR basically just surfaces the gui buttons into command line options so that scripts can use them.  Presently I have to maintain this all as a separate fork and retrofit it everytime you guys do an update.  It would be huge for us if we could get that capability included into the master branch.

simoninns, 2026-08-19:
How does this need to work though... the script starts the capture and the DdD app runs.  What stops the capture?  Can you walk me through the required interaction?  Also, is this purely CLI (no GUI pop) or is the GUI appearing not an issue?  I get the need to sync clocks, but I'm not clear on the actual scripting mechanics in use.

(to be clear - I'm super ok with including this and making sure it's in main - that's not the issue... I'm just not quite clear yet on how this is supposed to work)

closed 2026-08-20 by simoninns