Domesday86/DomesdayDuplicator #166: No clor in Ld analyse and after video export

didierroger1-lab, 2025-08-24:
Hi,
First of all thanks for your nice project. I use a CLD D925 to capture the RF flux. The player has been reviewed (new laser cartridge, power supply recapped). When I use the player on a TV I have a normal picture with the right color level. 
The capture with DdD works well with a signal level around 00,5/0,6.. But I can not get color when I use ld analyse or when I export the .tbc into a .mkv. When I use the vectorscope, I have just a point in the middle.
I. have check the cable used to transfer the signal, it is the right one (50 ohm cable) The level of signal is very good around 42 (for the black SNR). Any idea to recover the colors ?
Kinds regards

Didier

didierroger1-lab, 2025-08-24:
> That's very strange - if you're getting a successful RF decode at all,
> then you should be getting colour. Can you upload a short clip of one of
> your RF captures (say 50 frames - use something like ld-cut -s 1000 -l 50
> input.lds output.lds)? What disc are you capturing?

Hi
Thanks to your quick reply.I have not succeed to extract some frames of my
RF capture. It seems that I have many "mismatch" during the record with
message "player paused". I am afraid that it could be the cause of my issue.
The disc I try to capture is "Le bonheur est dans le prés" (PAL)

Kind regards

Didier

didierroger1-lab, 2025-08-25:
Hi again

I did some additional tests. In "Test mode", The recors stops after several
seconds with

didierroger1-lab, 2025-08-25:
After checking, the USB port is correct on my imac. For information, I use a windows 10 partition. May be a USB driver issue ? I attempt to extract something to show you.

didierroger1-lab, 2025-08-26:
File uploaded to you via google mail

simoninns, 2026-08-18:
Closing due to PR #172

closed 2026-08-18 by simoninns